w3c/magnetometer #1: Privacy implications?

chaals, 2016-07-12:
The fact that surrounding devices affect magnetic field may be an issue specific to this interface - in particular in providing location information for a previously calibrated area, or one in which a specific field pattern is created.

anssiko, 2017-10-05:
@lknik @alexshalamov can you investigate whether this should be noted in the spec, and see if appropriate mitigations and guidance to implementers are in place.

alexshalamov, 2017-10-05:
@anssiko We have identified similar attack vectors in [design document](https://docs.google.com/document/d/1Ml65ZdW5AgIsZTszk4mD_ohr40pcrdVFOIf0ZtWxDv0/edit#heading=h.lmg4m6asf9b4), need to think what mitigation strategies are applicable or whether we need to add new ones.

anssiko, 2017-10-05:
@chaals, submitted a PR https://github.com/w3c/magnetometer/pull/34 to fix this issue. PTAL.

anssiko, 2017-10-06:
@chaals, we landed https://github.com/w3c/magnetometer/commit/4c11c300c958e5094fd28f5377a73f89cb3decf6 to address this issue. If you have further comments or would like to amend the [Security and Privacy Considerations] section, please let us know.

Thank you for reviewing this specification, we appreciate your input.

[Security and Privacy Considerations]: https://w3c.github.io/magnetometer/#security-and-privacy

closed 2017-10-06 by anssiko

chaals, 2017-10-06:
@anssiko thanks. This is a big improvement. I think this is basically enough for now, although I may submit a PR that gives a small editorial tweak to clarify that there is important information in the generic API spec, if I get around to it...

lknik, 2017-10-07:
Hello,

Yeah this looks fine. I'm working on broader aspects but at the moment reliably can't tell when I finish.

commit 4c11c300 (2017-10-06): Fix #1: Expand Security and Privacy Considerations
commit f2d83583 (2017-10-06): Merge pull request #34 from w3c/security-privacy

Fix #1: Expand Security and Privacy Considerations