w3c/magnetometer #2: Security and Privacy Considerations

marcoscaceres, 2016-07-21:
It should be clear that there are none known *at this time* and the spec should encourage people report issues/concerns here on GH so that they may be included in the spec.

anssiko, 2017-02-01:
We're in process of getting the Security and Privacy story straight, and decided to have that discussion also for concrete sensor specs in the w3c/sensors repo to stay organized.

Currently, the Generic Sensor API captures the following known generic threats and discusses known mitigation strategies:
https://w3c.github.io/sensors/#security-and-privacy

The remaining open S&P issues are:
https://github.com/w3c/sensors/issues?utf8=%E2%9C%93&q=is%3Aissue%20is%3Aopen%20label%3Aprivacy%20label%3Asecurity

We'll waterfall sensor-specific considerations to the S&P sections of each concrete spec that extend the Generic Sensor API when done.

That said, I'll close this issues now, and encourage future feedback to the w3c/sensors repo.

lknik, 2017-02-03:
Hello,

I'll take care of security/privacy issues of magnetometer, as usual. In fact, I'm already onto that. Is there any deadline/preference?

anssiko, 2017-02-03:
Process-wise the spec should be feature complete when we enter Candidate Recommendation. As usual with standards, predicting timelines like that is not very useful thanks to consensus-driven process.

@lknik Your help is appreciated in closing the remaining security and privacy issues:

https://github.com/w3c/sensors/issues?utf8=%E2%9C%93&q=is%3Aissue%20is%3Aopen%20label%3Aprivacy%20label%3Asecurity

And if there are considerations specific to this Magnetometer spec not yet addressed in https://w3c.github.io/sensors/#security-and-privacy, I'd happily see a PR with your contributions that could be readily integrated.

lknik, 2017-02-07:
Thanks. Magnetometer definitely has its own issues, at least for "what can be done with it". I'll do it over the next few weeks (time-wise, quite constrained recently). Is there an estimated timeline? I think so far only one implementation exists.

lknik, 2020-06-12:
Hello

Nice [survey](https://dl.acm.org/doi/pdf/10.1145/3338498.3358650) with example interesting attacks using Generic Sensor APIs, including the option of detection of a site visited in another tab.

Do we want to update the considerations?

anssiko, 2020-06-12:
Referencing this paper from the security and privacy considerations would be helpful to inform implementers and allow them implement adequate mitigations. Feel free to submit a PR.

lknik, 2020-06-16:
I will compose something concise. In the meantime can we (temporarily) reopen this?

anssiko, 2020-06-16:
Thanks @lknik! Looking forward to your contribution.

lknik, 2020-06-18:
Here it is https://github.com/w3c/magnetometer/pull/56