w3c/magnetometer #3: Why magneticFieldX instead of x?

marcoscaceres, 2016-07-21:
Y u no just `.x` `.y` `.z` ? The event's `this` and current target are the MagnetometerSensor, so you don't need "magneticField" on the front ... or is there precedence/rationale for the long name?

riju, 2016-08-04:
[ Accelerometer](https://w3c.github.io/accelerometer/#accelerometersensorreading) and [Gyroscope](https://w3c.github.io/gyroscope/#gyroscopesensorreading) are also using the long name, similar to the platform api for [Windows](https://msdn.microsoft.com/en-us/library/windows/apps/windows.devices.sensors.magnetometerreading.aspx).

It does not hurt to be more verbose, but if you insist, we can make the name changes across the board.

marcoscaceres, 2016-08-04:
I know people got pretty upset about `querySelectorAll()` - so we like to keep things short where we can. 

I know it's bikeshedding, but the owner object already implies it's data type value.  @rwaldron, opinions (tm)? Shorten or keep verbose? What does J5 do?

anssiko, 2016-09-20:
F2F consensus, let's fix this.

rwaldron, 2016-09-20:
@marcoscaceres sorry I missed the original post. `Compass` & `Magnetometer` in J5 don't expose raw xyz magnetometer data (though they surely could), because generally the developer really just wants a heading in degrees, which is be computed from x and y.

```
let heading = Math.atan2(y, x);
let pi_2 = Math.PI * 2;

if (heading < 0) {
  heading += pi_2;
}

if (heading > pi_2) {
  heading -= pi_2;
}

let degrees = heading * (180 / Math.PI);
```

I view that as an undue burden on the developer. For true-north heading, just subtract a declination  angle. 

The **z** in x, y, z is used for computing the earth's magnetic field—I've written a robot that can navigate from any lat/lon point to any _other_ lat/lon point using a gps and compass/magnetometer—no **z** necessary.

rwaldron, 2016-09-20:
Re: Accelerometer and Gyro, the appropriate properties should be:

- Accelerometer: x, y, z (proper acceleration in terms of standard gravity or meters / s<sup>2</sup>; 1g = 9.8m/s<sup>2</sup>)
- Gyroscope: x, y, z (angular velocity in degrees / s)

rwaldron, 2016-09-20:
## Comparison, deriving heading in degrees: 

### MagnetometerSensor

```js
let sensor = new MagnetometerSensor();
sensor.start();

sensor.onchange = (event) => {
  let { magneticFieldX: x, magneticFieldY: y } = event.reading;
  let h = Math.atan2(y, x);
  let pi2 = Math.PI * 2;

  if (h < 0) {
    h += pi2;
  }

  if (h > pi2) {
    h -= pi2;
  }

  console.log(h * (180 / Math.PI));
};
```

### five.Magnetometer

```js
let sensor = new five.Magnetometer({ controller: "HMC5883L" });
sensor.on("change", () => {
  console.log(sensor.heading);
});
```

marcoscaceres, 2016-09-21:
Can we have J5's API please 🤗 The math makes me want to throw things.

anssiko, 2016-09-21:
We could have a high-level "Compass API" along the lines of J5 to go with the low-level Magnetometer. Actually, discussed this briefly at F2F yesterday.

@rwaldron How do J5 apps that require true north handle magnetic declination, should we try to bake that into the API so that the implementation can compensate, or leave it to the web developers to handle in script?

closed 2016-09-22 by riju

rwaldron, 2016-09-22:
> We could have a high-level "Compass API" along the lines of J5 to go with the low-level Magnetometer. Actually, discussed this briefly at F2F yesterday.

What would be the distinction? Why not just provide one: 

```
Magnetometer {
  heading,
  x,
  y,
  z,
}
```

> How do J5 apps that require true north handle magnetic declination

They would have to: 

1. Include a GPS to be of any use at all
2. Use lat/long to determine present local magnetic declination angle
3. Add or subtract the angle from the value of the `heading` property

Just now I ran a program that prints the heading in degrees as read from an HMC5883L, with the sensor on my desk. On the other side of my keyboard, I placed my iPhone with the compass app open and lined it up to north. Then I lined up the sensor's x axis to north (which was off). From there, I looked my present declination angle on http://www.magnetic-declination.com/ (at the moment, I'm in Boston), and it was -14 degrees. So my program looked like this: 

```js
const five = require("../");
const board = new five.Board();

board.on("ready", () => {
  const mag = new five.Magnetometer({
    controller: "HMC5883L"
  });

  mag.on("change", () => {
    console.log(Math.round(mag.heading - 14));
  });
});
```

I'm sure a phone or pc with gps could expose that data in an elegant way.

commit dd5dc625 (2016-09-22): Fixes #3