w3c/magnetometer #4: Model duplicates WebIDL

marcoscaceres, 2016-07-21:
> The Magnetometer Sensor’s associated Sensor subclass is the MagnetometerSensor class.
> The Magnetometer Sensor’s associated SensorReading subclass is the MagnetometerSensorReading class.

Both of the above are already stated in the webidl.

marcoscaceres, 2016-07-21:
Also, if there is other stuff in the model that is already in WebIDL, please remove that too.

anssiko, 2017-02-01:
We've settled on the following prose for model for concrete specs that subclass `Sensor`:
https://w3c.github.io/magnetometer/#model

The abstract `Sensor` baseclass' model is defined in:
https://w3c.github.io/sensors/#model

And its abstract operations at:
https://w3c.github.io/sensors/#abstract-operations

Thus I'll close this issue. Any generic [sic] feedback regarding the model, and its reuse by concrete sensors, or the use of WebIDL constructs should go to https://github.com/w3c/sensors/issues so that @tobie can fix/clarify them in that spec or in the WebIDL spec :-)

closed 2017-02-01 by anssiko