w3c/magnetometer #7: Is

marcoscaceres, 2016-07-21:
> To determine true north heading, add or subtract the appropriate declination angle.

Who should add/subtract the declination angle? The user agent? The developer? Won't hardware handle this?

marcoscaceres, 2016-07-21:
Same with:
> You can get real time value for magnetic declination from this website.

Who is "you" in the above?

marcoscaceres, 2016-07-21:
Same with:
> we need to apply various tilt compensation techniques for which we need the a 3-axis accelerometer.

Who is "we"?

anssiko, 2017-02-01:
Thanks, those were unclear indeed. Other than the last one had been fixed in a rounds of spec changes. The latest was just fixed in https://github.com/w3c/magnetometer/commit/a5a1a43eb9d7b168fbeec61047895a3561d7e104.

commit a5a1a43e (2017-02-01): Fix #7: Typos
closed 2017-02-01 by anssiko