w3c/magnetometer #16: Provide API to access uncalibrated magnetometer values

alexshalamov, 2016-11-03:
It would be good to have a way to get uncalibrated magnetometer sensor readings. One of the use cases is a magnetic button on VR enclosures for mobile device.

https://bugs.chromium.org/p/chromium/issues/detail?id=445926

anssiko, 2016-11-03:
I feel this alone is a strong enough use case that warrants us adding also uncalibrated magnetometer sensor readings to the [Magnetometer spec](https://w3c.github.io/magnetometer/).

To address this, I suggest we add a new constructor with a provisional name `RawMagnetometer`. @riju Could you craft a pull request we could review?

(For a background, I suggest you read the article by @borismus on the topic: http://smus.com/magnetic-input-mobile-vr/)

riju, 2016-11-03:
```webidl 
interface RawMagnetometer : Sensor {
  readonly attribute RawMagnetometerReading? reading;
};
```
option 1
```webidl 
interface RawMagnetometerReading : SensorReading {
    readonly attribute double x;
    readonly attribute double y;
    readonly attribute double z;
    readonly attribute double xBias;
    readonly attribute double yBias;
    readonly attribute double zBias;
};
```
Let's decide on the naming, and then I will put up the PR.

Ideally `xUncalib` would have been a good choice in naming, but going by the recent consensus, I think `RawMagnetometer` can signify that the readings are uncalibrated, so we don't need to use `xUncalib`, etc. 

Option 2

```webidl 
interface RawMagnetometerReading : MagnetometerReading {
    readonly attribute double xBias;
    readonly attribute double yBias;
    readonly attribute double zBias;
};
```
since we anyways don't distinguish between `x `and `xUncalib` in attribute name.

@tobie @alexshalamov  @pozdnyakov @anssiko  : chime in

tobie, 2016-11-03:
This is pretty cool.

Offering an option 3 which I think is acceptable in the calibration case (but not in the fusion case):

```webidl
[Constructor(optional MagnetometerSensorOptions sensorOptions)]
interface Magnetometer : Sensor {
  readonly attribute MagnetometerReading? reading;
};

dictionary MagnetometerSensorOptions : SensorOptions {
  boolean raw = false;
};

[Constructor(MagnetometerReadingInit magnetometerReadingInit)]
interface MagnetometerReading : SensorReading {
    readonly attribute double x;
    readonly attribute double y;
    readonly attribute double z;
    readonly attribute boolean raw;
};

dictionary MagnetometerReadingInit {
  required double x = 0;
  required double y = 0;
  required double z = 0;
  required boolean raw;
};
```

@rwaldron: thoughts? (background reading: http://smus.com/magnetic-input-mobile-vr/).

anssiko, 2016-11-03:
Option 3 looks nice, given this is indeed not sensor fusion.

Do we foresee a use case that warrants getting hold of both calibrated and raw readings via the same reading? If not, we're able to leave out option 1.

alexshalamov, 2016-11-04:
@tobis, @riju How do you plan to fit soft and hard iron bias in a single value?

alexshalamov, 2016-11-04:
@riju WDYT
```
interface RawMagnetometerReading : SensorReading {
  readonly MagnetometerReading softIron;
  readonly MagnetometerReading hardIron;
};
```

tobie, 2016-11-04:
> How do you plan to fit soft and hard iron bias in a single value?

I haven't looked at magnetometers sensors in details at all. Are those values provided by the sensor? The driver given some known HW data?

rwaldron, 2016-11-04:
@tobie 

> This is pretty cool.

+9001.

...

I read the background material and reviewed https://github.com/dodger487/MIST/tree/master/MIST/src/main/java/com/google/mist/magneto, and I think option 3 is ideal.

anssiko, 2016-11-04:
We seem to be converging toward option 3. Any concerns going with that?

pozdnyakov, 2016-11-04:
For uncalibrated magnetometer we need 6 reading values (3 extra values for biases along each axis), it seems option 3 does not suffice.

rwaldron, 2016-11-04:
Option 4?

```webidl
[Constructor(optional MagnetometerSensorOptions sensorOptions)]
interface Magnetometer : Sensor {
  readonly attribute MagnetometerReading? reading;
};

dictionary MagnetometerSensorOptions : SensorOptions {
  boolean raw = false;
};

[Constructor(MagnetometerReadingInit magnetometerReadingInit)]
interface MagnetometerReading : SensorReading {
  readonly attribute double x;
  readonly attribute double y;
  readonly attribute double z;
  readonly attribute double xBias;
  readonly attribute double yBias;
  readonly attribute double zBias;
  readonly attribute boolean raw;
};

dictionary MagnetometerReadingInit {
  required double x = 0;
  required double y = 0;
  required double z = 0;
  required double xBias = 0;
  required double yBias = 0;
  required double zBias = 0;
  required boolean raw;
};
```

When `raw` is `false`, the value of the *Bias properties would default to 0 and remain 0.

pozdnyakov, 2016-11-04:
`readonly attribute boolean raw;` at `Reading` is unneeded as it will contain the same value all the time for the same sensor: an uncalibrated sensor can never return calibrated readings and vice versa.

I would vote for the 1st option (let's split calibrated and uncalibrated sensors).

rwaldron, 2016-11-04:
> `readonly attribute boolean raw;` at `Reading` ...

meh, copy pasta. 

> I would vote for the 1st option (let's split calibrated and uncalibrated sensors).

An entire new class just for 3 additional values? I oppose this design suggestion.

riju, 2016-11-08:
@rwaldron : Windows does not have API for raw magnetometer readings. Also, we can expect developers to use calibrated readings much more than uncalibrated readings, in that case 3 doubles for bias and bool for raw is just waste of space for the object. If we can live with that, we will go with Option 4, else, I suggest Option 1.
If we split like option 1, we can use instanceof(sensor_var) also, maybe not that important.

riju, 2016-11-08:
@alexshalamov , I like the proposal @https://github.com/w3c/magnetometer/issues/16#issuecomment-258373915 , but the initialization for RawMagnetometerReading might look a bit weird.

tobie, 2016-11-08:
What precisely does the bias measure?

riju, 2016-11-08:
Bias = hard_Iron bias, distortion of magnetic field strength due to presence of magnet or similar material.
So [x,y,z] = magnetic field (without hard-iron compensation), and [x_bias, y_bias, z_bias] are the estimated hard-iron bias along the 3 axes.
We can then think, _uncalibrated = _calibrated + _bias. 
For the VR use case, presence of magnet will affect the bias values, and developer can take some action.

tobie, 2016-11-08:
What about:

```webidl

[Constructor(optional MagnetometerOptions magnetometerOptions)]
interface Magnetometer : Sensor {
  readonly attribute (MagnetometerReading or MagnetometerRawReading)? reading;
};

dictionary MagnetometerOptions : SensorOptions {
  boolean raw = false;
};

[Constructor(MagnetometerReadingInit magnetometerReadingInit)]
interface MagnetometerReading : SensorReading {
  readonly attribute double x;
  readonly attribute double y;
  readonly attribute double z;
};

dictionary MagnetometerReadingInit {
  required double x;
  required double y;
  required double z;
};

[Constructor(MagnetometerRawReadingInit magnetometerRawReadingInit)]
interface MagnetometerRawReading : MagnetometerReading {
  readonly attribute double xBias;
  readonly attribute double yBias;
  readonly attribute double zBias;
};

dictionary MagnetometerRawReadingInit : MagnetometerReadingInit {
  required double xBias;
  required double yBias;
  required double zBias;
};
```

tobie, 2016-11-08:
> For the VR use case, presence of magnet will affect the bias values, and developer can take some action.

So the bias values change over time? Where are those measurements taken from? The magnetometer itself?

riju, 2016-11-08:
> So the bias values change over time? Where are those measurements taken from? The magnetometer itself?

Yes. A magnet's presence affects magnetic strength, so bias changes.
Magnetometer itself provides the data, if it can.

tobie, 2016-11-08:
> Magnetometer itself provides the data, if it can.

So is there a reason not to provide this data all the time?

riju, 2016-11-08:
>So is there a reason not to provide this data all the time? (other than devx, e.g. perf) 

Ideally bias values are expected to jump as soon as the estimate of the hard-iron changes, and they should be stable the rest of the time. Depending on the use case, say for the VR use case, I guess bias data is needed all the time, to detect presence of magnet and how far it is from the sensor. If developer does not does not want bias values, he should use, Magnetometer interface, else he should use RawMagnetometer interface(if possible).

tobie, 2016-11-08:
So if the only reason we're creating a MagnetometerRawReading is to expose bias data, then maybe we should reconsider this.

That is: it might be useful to expose uncalibrated data and it might be useful to expose bias, but it's not clear to me that it's logical to expose bias on the uncalibrated sensor and not on the calibrated one also.

That's a broader conversation, though, that needs to happen at the generic sensor level.

Opening an issue there.

alexshalamov, 2016-11-15:
> What about option 3bis:

I think we should not derive uncalibrated reading interface from calibrated. This can produce unwanted side effects when developer checks what kind of reading is provided by sensor.

Omited constructors and init dicts to shorten example:

```
interface Magnetometer : Sensor {
   // Not needed, just to illustrate that the reading interface is changed
   // depending on the passed SensorOptions.
   readonly attribute (MagnetometerReading or MagnetometerRawReading) reading;
};

interface MagnetometerReading : SensorReading {
  readonly attribute double x;
  readonly attribute double y;
  readonly attribute double z;
};

interface MagnetometerRawReading : SensorReading {
  readonly attribute double x;
  readonly attribute double y;
  readonly attribute double z;
  readonly attribute double xBias;
  readonly attribute double yBias;
  readonly attribute double zBias;
};
```

Therefore:

```
rawMagn.reading instanceof MagnetometerRawReading; // true
rawMagn.reading instanceof MagnetometerReading; // false
```

@tobie What do you think?

rwaldron, 2016-11-15:
@alexshalamov I think that's reasonable

tobie, 2016-11-15:
Yeah. I don't really have an opinion, here. I'm surprised people still use instanceof, but that seems like a pretty big foot gun if they do, so I think it makes sense not to use inheritance here.

rwaldron, 2016-11-15:
> I'm surprised people still use instanceof, but that seems like a pretty big foot gun if they do, so I think it makes sense not to use inheritance here.

I just assumed he was using `instanceof` to illustrate that `reading` would be an instance of whichever, based on whatever the user set in the `Magnetometer` options.

tobie, 2016-11-15:
> I just assumed he was using instanceof to illustrate that reading would be an instance of whichever, based on whatever the user set in the Magnetometer options.

My understating is @alexshalamov was pointing out that with my proposed solution an uncalibrated reading would be `instanceof` both `MagnetometerReading` and `MagnetometerRawReading`.

rwaldron, 2016-11-15:
Oh, I see... yes. I completely misread/misunderstood—sorry about that!

riju, 2017-01-04:
forgot to link it to the [PR](https://github.com/w3c/magnetometer/pull/17/commits/f33d50c681882a97fbf58dbba0e9f4e84b02e2a0). 
Have a look.

anssiko, 2017-02-03:
This issue is still blocked on https://github.com/w3c/sensors/issues/151

Specific to magnetometer, @riju has an updated proposal in #21. 

Does the split into two top-level `Magnetometer` and `RawMagnetometer` interfaces with no new `SensorOptions` resonate with people? Is that a generic model that scales?

anssiko, 2017-05-04:
https://github.com/w3c/sensors/issues/151 was closed, unblocking this issue.

anssiko, 2017-05-11:
#21 landed after some serious discussion. Leaving this open to further comments.

tobie, 2017-05-11:
I don't understand why #21 was merged in the middle of huge conversation, with lot's of open issues left on it that are now not referenced from anywhere open in the tracker. This is not the first time this happens, see my review in w3c/ambient-light#26 that was not considered. I'm sorry maybe that makes me ubersensitive but I just find that hostile.

riju, 2017-05-16:
@anssiko : PTAL. 
Classify the magnetometer use cases into 2 buckets: 
- Use cases that require uncalibrated magnetometer
- Use cases that require magnetometer

The first 3 use cases in https://w3c.github.io/magnetometer/#usecases-and-requirements will benefit from Uncalibrated Magnetometer while the last 3 would need magnetometer.

In my view, 

- Metal(magnet) detection/ magnetic input / gesture recognition use cases in general will benefit from Uncalibrated Magnetometer reading. With uncalibrated magnetometer readings, there is no fear of calibration events (depending on platform and|or sensor manufacturer, can happen anytime nondeterministically) and because of that, we can reliably detect long presses and double clicks or identify presence of magnetized object in 3D space in relation to the device.

- Orientation/ Navigation (indoor)/ Compass heading can be achieved using (calibrated) Magnetometer readings.

anssiko, 2017-05-17:
Here's the synthesis I [promised] to deliver:

# Use cases

The Magnetometer specification documents [use cases and requirements]. Below is an evaluation of the use cases mentioned in the spec.

* There are [key use cases] for both (calibrated) magnetometer and uncalibrated magnetometer.
* For both the categories there exists native apps that make use of these capabilities.
* For uncalibrated magnetometer, there exists e.g. [various] [Metal Detector] apps, and a wide array of VR apps for Cardboard that utilize [Magnetic Input for Mobile VR]. Cardboard has [shipped over 10 million units] and as such Magnetic Input for VR can be consider a mainstream use case.
* Gesture recognition based on magnetic field is discussed in [Abracadabra], [Nenya], [MagGetz], and [MagiTact] papers. The [Magnetic Input for Mobile VR paper] states the VR use case can be implemented very reliably on (circa 2015) smartphones in contrast to more advanced gesture recognition techniques mentioned in other papers.

**Editor's conclusion: There are key use cases for both calibrated magnetometer and uncalibrated magnetometer, as well as various native apps that make use of both of these capabilities. This suggests we should consider uncalibrated magnetometer of similar importance to the existing (calibrated) magnetometer.**

# API design

Then, onwards to the API design discussion. Two proposals for the API shape were made:

## Separate constructors

Keep the current (calibrated) `Magnetometer`, and add a new `UncalibratedMagnetometer`. See the [proposed WebIDL].

The `Magnetometer` interface remains unchanged. For the newly added `UncalibratedMagnetometer` interface, the latest reading includes three entries whose keys are "x", "y", "z" and whose values contain [uncalibrated magnetic field] around the 3 different axes, and three additional entries whose keys are "xBias", "yBias", "zBias" and whose values contain the [hard iron distortion] correction around the 3 different axes.

Please note `Magnetometer.{x, y, z}` and `UncalibratedMagnetometer.{x, y, z}` values represent different concepts regardless of them sharing the same attribute names. As a thought experiment, consider the latter `UncalibratedMagnetometer.{a, b, c}`. We might consider renaming these attributes to make this clearer.

## Extend SensorOptions

Add an option to the `Magnetometer` constructor to turn it into an uncalibrated magnetometer. See the [proposed WebIDL for extending SensorOptions].

# API design: discussion

**The following technical details were shared in favour of separate constructors:**

* The API design should cater for varying magnetometer hardware built into devices. Separate constructors make it possible to expose the data from all three classes of magnetometer sensors shipping in devices, namely from magnetometer sensors that:
  * provide calibrated readings only
  * provide uncalibrated readings only, e.g. [Infineon 3D Magnetic Sensor]
  * provide both calibrated and uncalibrated readings

* Interfaces should clearly define model, operations and data that is provided by the implementation of the interface. If model, behavior or type of the data defined by the interface can be changed by construction parameter, it is a good design practice to split interface in favor of runtime variation. In this case, it is more obvious the meaning of `{x, y, z}` in each interface is different.

* Improves encapsulation. No need to dig into internals of the interface instance to understand what it is.

* Easier to manage different types of sensors in the future.

* Improves error handling and in our case, feature detection. If e.g. `UncalibratedMagnetometer` is not exposed on the window, then you know upfront it is not supported by the user agent.

* Works when a device has an uncalibrated magnetometer hardware only.

* Early web developer feedback received indicate developers found the options confusing.

* Consistent API shape. Currently all concrete sensors use the standard SensorOptions without sensor type-specific extensions. Not obvious to a web developer why the magnetometer should diverge from the otherwise consistent API shape.

**The arguments in favour of extending SensorOptions:**

* We've designed the Generic Sensor API around the idea of a 1:1 mapping between sensor type and WebIDL interface. Here, both of these options are actually variation of the same sensor. So from that perspective, it would make sense to treat them as the same interface.

* we're trying to be consistant across a complex landscape

* The right question to ask would be whether either API designs help surface useful information to developers, in a way that lowers cognitive load, makes them understand the overall system better, and thus be more effective. Or does it instead saw confusion.

* When we reason in terms of use cases, we should also reason in terms of how common those use cases are. Does it make sense to turn an API that's designed to help build experimental VR features into a full blown global variable, on par with orientation sensors or geolocation?

* One of the key raison d'être of the Generic Sensor API is to expose a coherent and consistent API for sensors and to help developers knowingly pick the right sensor for their particular use case. This implies having an understanding of the way sensors actually work. [...] So I don't think we should we really have the conversation about which API works better, but instead have the conversation about whether is it i [... cut ...]
[Magnetic Input for Mobile VR paper]: http://smus.com/magnetic-input-mobile-vr/mimvr-iswc-2015.pdf

[Infineon 3D Magnetic Sensor]: http://www.mouser.com/ds/2/196/Infineon-TLV493D-A1B6-DS-v01_00-EN-890494.pdf
[proposed WebIDL]: https://w3c.github.io/magnetometer/#api

[uncalibrated magnetic field]: https://w3c.github.io/magnetometer/#uncalibrated-magnetic-field
[hard iron distortion]: https://w3c.github.io/magnetometer/#hard-iron-distortion

[proposed WebIDL for extending SensorOptions]: https://github.com/w3c/magnetometer/issues/16#issuecomment-259090628

anssiko, 2017-10-05:
This issue has now been open for roughly five months, and the group has not come up with better names for the interfaces. This seems to indicate the group can live with the current names.

If this issue is now brought to someone's attention only by this act of closing it and that someone has a good proposal, s/he is encouraged to revisit this issue starting with the summary and set forth a concrete proposal to be considered by the group.

closed 2017-10-05 by anssiko