w3c/magnetometer #18: Shouldn't need Web Service to compute declination

mfoltzgoogle, 2016-12-26:
Is it feasible for the Web platform provide an API to compute this instead of relying on a Web service?  Or extend the reading to provide both raw and corrected values? It seems like something that most, if not all applications using the Magnetometer API would want.

Notes:
- Offline is a relevant use case for this API.
- The Web service in the example exposes the user's latitude and longitude over HTTP.

anssiko, 2017-02-01:
We're getting similar feedback from elsewhere too. This might warrant its own Compass API. Needs investigation.

@riju Could you investigate the status of the platforms APIs that provide magnetic declination.

anssiko, 2017-02-01:
Related #14.

anssiko, 2017-02-06:
I investigated this a bit.

Platform support: Android provides a [GeoMagneticField][2] class to get the magnetic declination. I haven't yet looked at other platform whether they expose something similar (contributions welcome!).

Platform support aside, I identified one potential issue: the declination angle could be used to validate whether the geolocation information (think IP geolocation and VPN use case) is accurate. Using a simple reverse lookup one could calculate whether the reported coordinates match the declination reported, see the [main field declination map][1]. That'd mean a user who might have a good reason to hide her geolocation, might inadvertently disclose that geolocation information reported is not accurate.

Couple of possible solutions in no particular order, with some pros and cons of mine:

* Define a standalone Compass API that requires UA to inform the user that when when access is allowed to the device's compass, also approximate location is disclosed in the process. Pros: Optimized for the arguably common use case. Cons: Possibly hard to communicate to the user the level of information disclosed (heading + approximate geolocation).

```
let sensor = new Compass();
sensor.start();

sensor.onchange = () => {
    console.log("Heading " + sensor.heading);
};
```

* Add a `getDeclination()` convenience method on `Magnetometer` that takes latitude, longitude, and altitude as arguments, and returns the magnetic declination (in degrees) at the given location (here sync API for brevity). The web developer is responsible for acquiring the geolocation of the user, e.g. using IP geolocation, Geolocation API that relies on the device's GPS, or other means. Pros: This is basically the web service provided by NOAA but implemented natively, supports offline use. Cons: ?
```
var magneticNorthDegrees = Math.atan2(sensor.y, sensor.x) * (180 / Math.PI);
var declinationDegrees = sensor.getDeclination(lan, long);
var trueNorthDegrees = magneticNorthDegrees + declinationDegrees;
```

* Leave all of this to the web developer, as currently specified, see [example][3]. Pros: less API surface, cons: no support for offline.

cc @alexshalamov @riju @lknik for further comments, and ideas.

(@mfoltzgoogle The web service used in the [example][3] should obviously use https for sensitive information such as geolocation coordinates. Fixed the example, since NOAA seem to have updated their service similarly recently, and finally enabled https.)

[1]: https://upload.wikimedia.org/wikipedia/commons/thumb/1/11/World_Magnetic_Declination_2015.pdf/page1-1599px-World_Magnetic_Declination_2015.pdf.jpg

[2]: https://developer.android.com/reference/android/hardware/GeomagneticField.html

[3]: https://w3c.github.io/magnetometer/#example-721a04cc

anssiko, 2017-10-05:
Addressing comments:

>Offline is a relevant use case for this API.

The Geolocation API exposes [heading] "in degrees, where 0° ≤ heading < 360°, counting clockwise relative to the true north." The heading is also exposed similarly through the Generic Sensor-based [Geolocation Sensor] for a more modern API shape, under incubation.

The offline requirement is satisfied by the existing Geolocation API.

>The Web service in the example exposes the user's latitude and longitude over HTTP.

This has been since fixed.

I'll close this issue now, we should reopen if new information resurfaces.

[heading]: https://dev.w3.org/geo/api/spec-source.html#heading
[Geolocation Sensor]: https://anssiko.github.io/geolocation-sensor/

closed 2017-10-05 by anssiko