w3c/magnetometer #19: Local coordinates could use further detail/examples

mfoltzgoogle, 2016-12-26:
It's not clear what local coordinates means for:
- Tablet or convertible devices that are used in both portrait and landscape.  Do coordinates change with screen orientation?
- Smart watch form factor.
- Maker-kit-style offboard sensors, e.g. [1].  This may just be "implementation dependent" :-)

[1] https://www.adafruit.com/product/2021

anssiko, 2017-02-01:
We discussed whether to add more examples of device classes to the spec beside a typical mobile device. Since we cannot anticipate all future form factors some of them would always be implementation dependent, as you note.

The current normative prose states "local coordinate system defined by the device" that is to say it is implementation dependent, and mirrors the conventions of each platform and its corresponding underlying APIs. The currently shipping Device Orientation API is similarly vague in this respect. We'll clarify https://w3c.github.io/magnetometer/#local-coordinate-system along those lines, when we figure out exactly what type of text would be helpful.

Our primary focus has been to enable this API on mobile devices first, so we haven't explored alternative classes of devices in detail yet.

anssiko, 2017-10-05:
Thanks for the feedback @mfoltzgoogle, sorry that it took so long to respond.

>Tablet or convertible devices that are used in both portrait and landscape. Do coordinates change with screen orientation?

The hardware sensor has predefined axes, and as such, if the screen orientation changes, the axes won't, since the physical assembly of the sensor does not change on screen orientation change.

We will investigate this further in https://github.com/w3c/sensors/issues/257 from the API ergonomics point of view, this is a generic issue, tagged as Level 2.

>Smart watch form factor.

In this form factor it will be most probably like in the mobile device form factor, defined by the manufacturer of the smart watch, z pointing upwards from the screen etc.

>Maker-kit-style offboard sensors, e.g. [1]. This may just be "implementation dependent" :-)

Yes, this is implementation dependent. This is the reason we use "a device that hosts the sensor" language in related specs.

>Regarding the orientation question, based on the text in the spec, it seems like a mobile device implementing a compass would need to listen to Device Orientation to know how to translate inputs from the sensor onto the orientation of the rendering.

Similarly in scope for https://github.com/w3c/sensors/issues/257

I will close this issue now, but should you have any further comments we can open this issue again.

closed 2017-10-05 by anssiko