w3c/magnetometer #24: Add use-cases section

alexshalamov, 2017-04-21:
Would be nice to add use-cases section, I have few in mind:

- Metal detector web app
- Providing input using magnet button for VR enclosures (e.g., vrbutton.appspot.com)
- User-specific manual fusion with other low-level sensors

riju, 2017-04-21:
>Providing input using magnet button for VR enclosures (e.g., vrbutton.appspot.com)
 https://bugs.chromium.org/p/chromium/issues/detail?id=445926

riju, 2017-04-21:
Indoor Navigation System may use magnetometer on mobile phones. For example, IndoorAtlas (http://www.indooratlas.com/how-it-works/)
Use cases (Proximity Advertising, Way-finding in malls or airports Geo-Fencing, etc)

alexshalamov, 2017-04-21:
Calculating compass heading is another simple use-case

PR #25 (2017-04-25): Add use cases section
Add use cases for magnetometer.
This is an intial list of use cases.
We can add more use cases later.

Fixes #24

***
[Preview](https://s3.amazonaws.com/pr-preview/riju/magnetometer/addUseCases.html) | [Diff](https://s3.amazonaws.com/pr-preview/w3c/magnetometer/4484e03...riju:88e2dd4.html)
commit 88e2dd4b (2017-04-25): Add use cases and requirement section.

Add use cases for magnetometer.
This is an intial list of use cases.
We can add more use cases later.

Fixes #24

Also, make biblio.json inline
closed 2017-04-25 by riju