w3c/magnetometer #59: Request for Magnetometer web developer feedback

anssiko, 2021-11-05:
>RESOLUTION: Update Status of the Document to indicate Magnetometer is looking for developer feedback and high value use cases

via https://www.w3.org/2021/10/29-dap-minutes.html#t08

### Call for web developer feedback

📢 Web developers! To help accelerate the spec advancement, please share your use cases that require Magnetometer as well as any pointers to web experiments or prototypes using the Magnetometer. You can simply drop a comment into this issue.

Your contributions will be acknowledged in the update to the [use cases](https://www.w3.org/TR/magnetometer/#usecases-and-requirements) documented in the spec and will help browser implementers prioritize this API accordingly.

### How to enable the Magnetometer API for experimentation

Enable `chrome://flags/#enable-generic-sensor-extra-classes` flag in Chromium-based browsers. For all supported browsers and versions, see https://caniuse.com/magnetometer

### Spec

https://w3c.github.io/magnetometer/

### Demos

https://github.com/intel/generic-sensor-demos

### Developer docs

https://web.dev/generic-sensor/
https://w3c.github.io/motion-sensors/
https://developer.mozilla.org/docs/Web/API/Magnetometer

Zibri, 2023-01-22:
I enabled the API but I think the magnetometer api should be like the movement (accelerometer/gyro) api.

and it need at least 60 updates per second (like the other APIs) to be useful for a wider range of application aside from finding metals and "north".

anssiko, 2023-01-23:
@Zibri thanks for your interest toward this API. What are your specific use cases that need at least 60 updates per second?

As you probably noticed, the maximum sampling frequency is limited to mitigate possible threats discussed in https://www.w3.org/TR/magnetometer/#security-and-privacy This API needs to strike a balance between functionality, privacy and security. Strong real-world use cases would help motivate work on solutions that'd allow relaxing this limitation while preserving the privacy and security characteristics of the API.

You can read more about security and privacy considerations for sensor APIs in general
at https://www.w3.org/TR/generic-sensor/#security-and-privacy

Looking forward to your use case contributions!

Zibri, 2023-01-23:
Threats? I don't agree (I'm a security consultant) I think authorization should be used like camera/microphone and then be free to experiment with it, allowing to use a PWA instead of writing a normal application.

Off topic:
another annoying thing of PWAs is that the Audio sampling rate seems limited to 48khz while my phone and os support 96khz too. If I use "Spectroid" app I can set the sampling rate to 96khz and it works beautifully since the mic and speaker support ultrasounds, but not in a PWA..